OpenEarable/app #1: posture tracker application example

TobiasRoeddiger, 2023-09-01:


DennisMoschina, 2023-09-09:
"I created a quick mockup sketch for an idea on how the head could be rendered. The range of motion in this model is limited, as it is not a real 3D model, but I believe it should be suitable for real-world applications.

[image]

TobiasRoeddiger, 2023-09-09:
the sound ore liberty buds have a really nice ui for this i‘ll send it through when i‘m back

DennisMoschina, 2023-09-09:
I found this online. I'll implement something similar to this

[image]

TobiasRoeddiger, 2023-09-09:
exactly, that‘s what i meant

TobiasRoeddiger, 2023-09-11:
very nice, does it work with OpenEarable?

closed 2023-12-05 by TobiasRoeddiger