OpenEarable/app #55: after disconnecting from an earable, it is not possible to reconnect

TobiasRoeddiger, 2024-01-23:
connect to OpenEarable, then tap on it to disconnect. it is no longer possible to connect again

---
reported by @kkai on iOS

o-bagge, 2024-01-25:
The problem is with the _retryConnection method in open_earable_flutter (ble_manager.dart line 65).  I moved the recursive function call into the switch case and it seems to solve the problem. Why do we need multiple retries @DennisMoschina ?

```dart
StreamSubscription? _retryConnection(int retries, DiscoveredDevice device) {
    if (retries <= 0) {
      _connectingDevice = null;
      return null;
    }
    return _flutterReactiveBle.connectToAdvertisingDevice(
        id: device.id,
        prescanDuration: const Duration(seconds: 1),
        withServices: [sensorServiceUuid]).listen((event) async {
      switch (event.connectionState) {
        case DeviceConnectionState.connected:
          _connectedDevice = device;
          _flutterReactiveBle.requestMtu(deviceId: device.id, mtu: mtu);
          if (deviceIdentifier == null || deviceFirmwareVersion == null) {
            await readDeviceIdentifier();
            await readDeviceFirmwareVersion();
          }
          _connectionStateController.add(true);
          _connectingDevice = null;
          return;
        case DeviceConnectionState.disconnected:
          _connectedDevice = null;
          _connectingDevice = null;
          _deviceFirmwareVersion = null;
          _deviceIdentifier = null;
          _connectionStateController.add(false);
          _connectionStateSubscription = _retryConnection(retries - 1, device);
        default:
      }
      // _connectionStateSubscription = _retryConnection(retries - 1, device);  <---- moved this line up
    });
  }
```

DennisMoschina, 2024-01-25:
> Why do we need multiple retries 

On Android we somehow could only connect after multiple retries so I added the `_retryConnection` method as a quick fix

o-bagge, 2024-01-27:
Ok makes sense. Do you think it will still work on android with this change?

TobiasRoeddiger, 2024-10-15:
@cadivus is this problem resolved?

closed 2025-01-27 by TobiasRoeddiger